ComThings/PandwaRF #4: Pandwarf not working on Galaxy S7 + android 6.0.1 ($ rm -Rf /*)

trollwookiee, 2017-01-17:
Hey I don't seem to be able to make Pandwarf work on my phone: Galaxy S7 + android 6.0.1 // for example the spectrum analyser doesn't show graph and when it does, the sampling seems to stop after one second and the entire app gets irresponsive. Then i'm forced to quit and reinit the app. I tried to wipe cache, and reset config, but still no luck. - Advice? -

Galaxy S7 + android 6.0.1

PsychoBitchy, 2017-01-17:
Some more details:
I'm able to make the SpecAn to work by flushing app data + cache.
However it displays a very slow data rate.
If I stop it / reset it / or change RF band scale, the data rate drops to 0 Meas/s and the app displays no measurements until I flush cache / app data.

[image]

[image]

[image]

[image]

[image]

[image]

PsychoBitchy, 2017-01-17:
I also stumbled upon an erratic behaviour on the SpecAn screen: sometime just swiping over the graph results in the frequency graph being originated at 0 mhz:
[image]

trollwookiee, 2017-01-17:
Thanks.  
If you stick to a static setting when doing a Specan, does it work ? 
Please also check https://github.com/ComThings/PandwaRF/wiki/Support-Known-Issues#spectrum-analyzer-doesnt-support-slidingzoomingpinching

We have to rework the specan UI. This is scheduled for march.

Besides of Specan, what are the other errors you have ? Disconnect ? RX errors ?

PsychoBitchy, 2017-01-18:
If by " static setting " you're meaning " flush cache, reboot then don't touch anything on default setting, yes, well sort of: I hadn't been able to locate any keyfob TX with this, but it displays a graph.  
Moreover it feels like the app is almost sucking all my device CPU, as the home button becomes laggy.

duan78, 2017-02-09:
Same problem here (SGS 7 Edge)
--> SpecAn freeze (or blank after stop/reload)

trollwookiee, 2017-04-19:
High CPU usage has been fixed in 1.236. Please check.

PsychoBitchy, 2017-04-24:
Well, sorry to bring bad news. 
In the meantime s7 went on Android 7.  
This time, with last firmware, SpecAn doesn't show at all.
[image]
[image]
[image]

PsychoBitchy, 2017-04-25:
Hello.  
I have been able to finally get it to work and it's a marvel! 
The specAn however sometimes stops working and I have to kill the background paused thread to get it again.  
There's still a somewhat unreliable worker around there but nothing rendering the app unusable.

You can close that thread. 
Thank you for your dedication.

trollwookiee, 2017-04-26:
Great !

closed 2017-04-26 by trollwookiee

alastacat, 2017-10-25:
I seem to be having the exact same problem using my pixel 2 using android 8.0.0.  Software version 0.9.17 and Firmware version 1.0.17.
Flushing app data + cache makes the spectrum analyser run, but the data rate drops to 0 Meas/s and not picking up signal. 
Next time I run the spec analyser it doesn't start until flush again.  
Is there any detail on how this was fixed?

trollwookiee, 2017-10-26:
On some phones, Specan causes issues. 
In general, SpecAn has problems to restart. This is cause by some background tasks not properly cleaned.
I am investigating.

trollwookiee, 2017-10-26:
If issue persists, please open a new defect.