ComThings/PandwaRF #12: Add support for 4-FSK to decode DMR

jthenley, 2017-03-01:
Please can you add support for 4-FSK demodulation for CC1111 (like YARS Stick One) in order to capture DMR data.

trollwookiee, 2017-03-01:
We can try it. But I don't have DMR device to test it.
If we enable 4-FSK it will be totally blindly on our side. Will you be able to test it and provide feedback ?

jthenley, 2017-03-01:
Yes absolutely.

trollwookiee, 2017-11-30:
Fixed

closed 2017-11-30 by trollwookiee

phretor, 2018-04-17:
Is this implemented at the moment?

```
   1347     def reprMdmModulation(self, radiocfg=None):
   1348         mod = self.getMdmModulation(radiocfg)
-> 1349         return ("Modulation:          %s" % MODULATIONS[mod])
   1350
   1351     def getMdmChanSpc(self, mhz=24, radiocfg=None):

KeyError: 64L

In [5]: MODULATIONS
Out[5]:
{0: '2FSK',
 8: '2FSK/Manchester encoding',
 16: 'GFSK',
 24: 'GFSK/Manchester encoding',
 48: 'ASK/OOK',
 56: 'ASK/OOK/Manchester encoding',
 112: 'MSK',
 120: 'MSK/Manchester encoding'}
```

trollwookiee, 2018-04-17:
It should be right now. Please check https://github.com/ComThings/PandwaRF/commit/1fa87e976dc5932b41279c2d726ea1fd466685f9

phretor, 2018-04-18:
I'm glad I've ended up implementing the same change. Thought it was more complicated, but after reading the CC data sheet, the change became obvious. Thanks a lot for the quick response.

So far, I've tested it on a Mac, but I'm assuming RFCat is not meant to work on macOS. I'll let you know how it goes under Linux.

trollwookiee, 2018-04-18:
Ok thanks

phretor, 2018-04-26:
I'm not sure I understand this 100%. I guess this doesn't automatically grant support for 4FSK, right? I don't have the PandwaRF firmware source code to check, but RFCat's firmware for the CC1111 doesn't seem to set the registers for 4-FSK: https://github.com/atlas0fd00m/rfcat/blob/master/firmware/include/cc1110-ext.h#L1061 - am I missing something?

trollwookiee, 2018-04-26:
This is not the FW we use for RF.
Did you look at the video in https://pandwarf.com/news/pandwarf-demo-capture-retransmit-data-several-modulations/
At the end I make a 4FSK transmission.
Is it what you want ?