ComThings/PandwaRF #14: Brand new Pandawarf all leds on

gas17gas, 2017-06-19:
I unpacked my new pandawarf, I connected a 868MHz antenna and an usb cable to power it, all leds still on, nothing happen, reset button do nothing, impossible to put the device in recovery mode, bluetooth devices are not detected, nothing works.
Can I try something to fix this problem?

trollwookiee, 2017-06-20:
Can you tell me what is the revision marking on the PCB ?
Does the Blue LED or orange LED blink ?
Were you able to connect it once ?
Please send us an email at pandwarf@comthings.com to continue investigation

trollwookiee, 2017-06-27:
it looks like there is no FW flashed...
If you ordered a PandwaRF Bare without FW, please be careful:
PandwaRF Bare are **development boards**. You need to **flash your own FW** (Nordic and CC1111).

closed 2017-06-28 by trollwookiee