ComThings/PandwaRF #18: RfCat RFxmit function gives the error "ChipconUsbTimeoutException: Timeout waiting for USB response

trollwookiee, 2017-09-08:
when i try to send some data the program show me the following error: >>> d.RFxmit('blahblahblah')
Traceback (most recent call last):
File "<console>", line 1, in <module>
File "/home/maverick/pandwarf/PandwaRF/SW/rfcat/rflib/chipcon_nic.py", line 226, in RFxmit
self.send(APP_NIC, NIC_XMIT, "%s" % struct.pack("<HHH",len(data),repeat,offset)+data, wait=wait)
File "/home/maverick/pandwarf/PandwaRF/SW/rfcat/rflib/cc1111client.py", line 851, in send
return self.recv(app, cmd, wait)
File "/home/maverick/pandwarf/PandwaRF/SW/rfcat/rflib/cc1111client.py", line 815, in recv
raise(ChipconUsbTimeoutException())
ChipconUsbTimeoutException: Timeout waiting for USB response.

trollwookiee, 2017-09-08:
It looks like RfCat enters RX mode straight after you enter research mode. So the RFxmit command is received while in RF mode, and RFxmit ack doesn't come back to PC.
This causes the timeout issue.

So before calling d.RFxmit(), just call d.setModeIDLE().

Sequence is :     
d.setFreq(433000000)
d.setMdmModulation(MOD_ASK_OOK)
d.makePktFLEN(250)
**d.setModeIDLE()**
d.RFxmit("HALLO")

closed 2017-09-08 by trollwookiee

ssernabeu, 2019-07-01:
I get the following error when I try to transmit after receiving. Regardless of putting **d.setModeIDLE()** before or not. Has anyone managed to overcome this? I have not seen any solutions online...

Traceback (most recent call last):
  File "/home/pi/testing/receive.py", line 202, in <module>
    d.setModeIDLE()
  File "/usr/local/lib/python2.7/dist-packages/rflib/chipcon_nic.py", line 335, in setModeIDLE
    self.setRfMode(RFST_SIDLE)
  File "/usr/local/lib/python2.7/dist-packages/rflib/chipcon_nic.py", line 313, in setRfMode
    r = self.send(APP_SYSTEM, SYS_CMD_RFMODE, "%c" % (self._rfmode) + parms)
  File "/usr/local/lib/python2.7/dist-packages/rflib/chipcon_usb.py", line 700, in send
    return self.recv(app, cmd, wait)
  File "/usr/local/lib/python2.7/dist-packages/rflib/chipcon_usb.py", line 664, in recv
    raise(ChipconUsbTimeoutException())
rflib.chipcon_usb.ChipconUsbTimeoutException: Timeout waiting for USB response.

I have tried this on 2 different YS1 with different firmware versions(0543 and 5535), compiler SDCCv350 and rflib 464.

SorenAndreasen, 2019-07-04:
i get this very similar error when trying to call:
python rfcat-rolljam.py -f 433900000 -r 1818 -m -40 -o -2500000 -O capture.io -k
Configuring Scanner on Frequency: 433900000

then it gives me 
Configuring Scanner on Frequency: 433900000
Traceback (most recent call last):
  File "rfcat-rolljam.py", line 55, in <module>
    d = RfCat(idx=0)
  File "/home/pi/Downloads/PandwaRF-master/SW/rfcat/rflib/chipcon_nic.py", line 149, in __init__
    GollumDongle.__init__(self, idx, debug, copyDongle, RfMode)
  File "/home/pi/Downloads/PandwaRF-master/SW/rfcat/rflib/cc1111gollum.py", line 80, in __init__
    self.gollumRfConfigure()
  File "/home/pi/Downloads/PandwaRF-master/SW/rfcat/rflib/cc1111gollum.py", line 86, in gollumRfConfigure
    self.setAmpMode(RF_POWER_AMPLIFIERS_ACTION_ALL_OFF)
  File "/home/pi/Downloads/PandwaRF-master/SW/rfcat/rflib/cc1111gollum.py", line 197, in setAmpMode
    self.send(APP_RF, SYS_CMD_RF_TXRX_POWER_AMP_ACTION, "%c" % ampmode)
  File "/home/pi/Downloads/PandwaRF-master/SW/rfcat/rflib/cc1111client.py", line 870, in send
    return self.recv(app, cmd, wait)
  File "/home/pi/Downloads/PandwaRF-master/SW/rfcat/rflib/cc1111client.py", line 834, in recv
    raise(ChipconUsbTimeoutException())
rflib.cc1111client.ChipconUsbTimeoutException: Timeout waiting for USB response.

i'm using 2 Yard Stick Ones on a raspberry pi. they have the latest rfcat firmware, I'm using the script from here https://github.com/ComThings/rfcat-rolljam  the firmware i installed was from the official rfcat repository. is there something specific i'm doing wrong ?

SorenAndreasen, 2019-07-04:
are yard stick one's not supposed to work with this ?

ssernabeu, 2019-07-04:
Hi Soren,

First, I believe the function d = rfcat() does not accept any inputs so perhaps try to take idx=0 out?

I solved the errors as of yesterday. I was trying to run a listen and transmit script on a loop. Some conclusions I have come to are: the chipcon error is always expected to appear when using RFrecv(time_in_miliseconds) if nothing is heard so use a try-except to pass the error so your program does not crash. When something was heard there seemed to be no problem at all! I also got the chipcon error when listening and then trying to transmit but it seems this was caused by the function d.cleanup() so if you are using this you are gonna want to either delete it or put it at the very end of the script, outside of any while loops. I thought it only emptied variables and queues but I probably missed something. 

I can't really tell you more without seeing your script.

Sergio