ComThings/PandwaRF #19: ER:GPIOs for RollJam

DAVID9SE, 2017-09-09:
It would be great to use the gpios using Javascript just for a 1 euro upgrade to my pandwa there would be a way to hack Rolling codes.I know that you are trying to make pandwaRF hack rolling codes so i think this idea can make it really cheap. The idea is that i would connect a 1EURO 433mhz OOK transmitter to the gpios and that would allow us to jam and at the same time listen for the data.

For people who don't understand something here are all the links.
433 transmitter:
https://www.amazon.com/433Mhz-Transmitter-Receiver-Link-Arduino/dp/B016V18KZ8/ref=sr_1_4?s=electronics&ie=UTF8&qid=1504963252&sr=1-4&keywords=433mhz+transmitter
RollJam;
https://makezine.com/2015/08/11/anatomy-of-the-rolljam-wireless-car-hack/
Thanks.
PS:Sorry for my English ,but i am from Europe so i am not the best.

DAVID9SE, 2017-09-10:
EDIT:
1.Problem
I am not even able to replay my key:
If i sniff my key (of course not near the car so the car can not recieve the code.)and than xmit next to the car nothing happens.
My settings:
433.8 mhz
5000/3600/19607 bts
1000 bytes
33/14/90 bytes

trollwookiee, 2017-09-11:
GPIO: ok, I will look at it.
TX problem: Please move this to our forum: http://pandwarf.boards.net/

This Github Issue tracker is for enhancement or reporting bugs.

DAVID9SE, 2018-01-20:
Hi, I would like to ask if there is a progress with the gpios (If it is possible).
Thanks.

EvilDonkey420, 2019-02-13:
For an easy quick fix just use python and a yardstick one

trollwookiee, 2019-02-19:
@DAVID9SE this felt into the bottom of my priority queue :-(
The problem with this feature is the BLE delay between the Smartphone and PandwaRF.
The GPIO TX part is not really complex.
For GPIO RX it is more difficult.
We have 4 GPIOs available:
- BT_PORT_10
- BT_PORT_14
- BT_PORT_16
- BT_PORT_18
So I guess we can come with an Android/JS API to send data to a given GPIO.
I will keep you posted.

askumar101, 2019-04-01:
@DAVID9SE I agree, in Sept last year they boasted rolling code support early 2019; so i wasted $1000 on a Rogue Pro in anticipation.......ffs its April 2019 and still waiting, and no further updates, now iv utilized other avenues; i dont believe it will happen anytime soon.

DAVID9SE, 2019-04-01:
@askumar101 happy that someone agrees with me.I bought my pandwa more than a year back and the only thing I was able to do is jam and capture static codes.I think I should be able to do more with a 100$+ device. I hope that there will be gpio in the future so I can upgrade it.

asiudoisjdfo, 2019-06-21:
Couldnt agree more

DAVID9SE, 2019-10-16:
Hi I guess this is the end. I see that the PandwaRF classic is now discontinued. I had hopes for this product and I think this is a real stab in the back. There was a lot of stuff promised and now it will be only on the rogue. I just wanted to put out here that if there is anybody interested in making a custom firmware I would be more than happy to join them. Thanks a lot

trollwookiee, 2019-10-21:
Closed as out of scope. GPIOs will not likely be exposed to Javascript API.

closed 2019-10-21 by trollwookiee

askumar101, 2019-10-21:
Indeed its a stab in the back, that's all this product has ever been due to its management who now realize there's more money in more expensive variants, hence all the promises made are long forgotten, with the efficiency of an unloaded gun.

DAVID9SE, 2020-08-24:
@askumar101 After a year we can see now it is totally dead. Anyways @askumar101 could you please contact me on my email(you don't have yours on GitHub.) I would like to ask some stuff regarding rf hacking.

askumar101, 2020-08-26:
@DAVID9SE my email is now visible...