ComThings/PandwaRF #24: Bluetooth LE services and characteristics documentation

ptigas, 2017-10-17:
Hi,

I am interested in developing a mobile application using PandwaRF and would be great if I could use directly Bluetooth LE services and characteristics.

Could you please provide some documentation? Also, are there any plans open sourcing the data processing part or it's proprietary?

p.

trollwookiee, 2017-11-13:
Documention effort for BLE services and characterisitics is ongoing. ETA december 2017.

kevanlanister735, 2019-06-19:
Still ongoing??? If this was documented, we would also be able to use BLE Services from Web Bluetooth?

trollwookiee, 2019-06-24:
First draft of BLE services documentation have been added to wiki here: [PandwaRF BLE Services & characteristics](https://github.com/ComThings/PandwaRF/wiki/Developer-BLE-Services-Characteristics).

This is OK to start with. What remains to be described is

- RfCat messages description
- Mapping of RfCat messages onto BLE NUS
- RfCat message segmentation process

closed 2019-08-02 by trollwookiee