ComThings/PandwaRF #26: Unable to do RX/TX - RX Setup takes long time

ryan-dsouza, 2017-10-25:
Bluetooth connects fine. However unable to receive an RF signals as RX setup seems to go on forever. 
Attached logs
Phone is MotoG
Android ver 5.1

Tried with smaller buffer lengths. Any help will be appreciated.

PandwaRF Device Information:
MAC Address:                  E6:C2:E5:4B:62:6C
FW CC1111 Commit Id:          
FW CC1111 Version:            1.0.17
FW Nordic Commit Id:          
FW Nordic Version:            0.9.17

PandwaRF App Information:
APPLICATION_ID:               com.comthings.pandwarf
BUILD_TYPE:                   release
FLAVOR:                       pub
VERSION_CODE:                 10254
VERSION_NAME:                 1.2.54
App Commit Id:                b877d29
Gollum Android Lib Version:   0.7.17-pub-release
Gollum Android Lib Commit Id: b877d29
Gollum Device Lib Version:    0.0.7-pub-release
Gollum Device Lib Commit Id:  e340293
Gollum Device Lib Flavor:     e340293

Phone Information:
serial:                    TA9290DPBV
bootloader:                0x411A
manufacturer:              motorola
brand:                     motorola
hardware:                  qcom
board:                     MSM8226
device:                    falcon_umts
model:                     XT1032
display:                   LPB23.13-56
id:                        LPB23.13-56
type:                      user
user:                      hudsoncm
host:                      ilclbld36
fingerprint:               motorola/falcon_retgb/falcon_umts:5.1/LPB23.13-56/60:user/release-keys
tags:                      release-keys
time:                      1439854829000
radio firmware:            MSM8626BP_1032.3116.98.00R EMEA_CUST
version.codename:          REL
version.release:           5.1
version.incremental:       60
version.sdk_int:           22
Hash: cd8403413815ca03f9abc46ebfba33a7[
[logcat.zip](https://github.com/ComThings/PandwaRF/files/1415205/logcat.zip)
](url)

trollwookiee, 2017-10-25:
Hi, can you enable Dev Mode and go to CC1111 RF page ?
Then click on Read all.
Are registers returned ?

ryan-dsouza, 2017-10-25:
Yes I did that - None of the registers are returned - the page is blank

rfenthusiast, 2017-10-25:
[image]

ryan-dsouza, 2017-10-25:
In the log file the functions are called getC1111RfRegisters and getC1111RfRegisters done.

ryan-dsouza, 2017-10-25:
[image]

trollwookiee, 2017-10-25:
I looked at the trace, the RF link seems dead:
cc1111_ReadReg: Error, peek returned null
rfsub1g_cc1111_Rf_GetMdmDRate: Failed, readburst returned null
...

Also I don't know the cause since I don't have the beginnign of the connection.
I suggest that you:

1. physically reset the PandwaRF
2. kill the Android app
3. restart the app and connect to PandwaRF
4. don't do RX capture, but go directly to CC1111 RF and try to read registers
5. send me the logcat debug trace.

thanks

ryan-dsouza, 2017-10-25:
Thanks for the quick response - Suspected the RF  ( seems to be the same problem on another 
device as well - Not sure what caused it - It did work a few times). 
------------------
[logcat-1.zip](https://github.com/ComThings/PandwaRF/files/1415719/logcat-1.zip)

PandwaRF Device Information:
MAC Address:                  E6:C2:E5:4B:62:6C
FW CC1111 Commit Id:          
FW CC1111 Version:            1.0.17
FW Nordic Commit Id:          
FW Nordic Version:            0.9.17

PandwaRF App Information:
APPLICATION_ID:               com.comthings.pandwarf
BUILD_TYPE:                   release
FLAVOR:                       pub
VERSION_CODE:                 10254
VERSION_NAME:                 1.2.54
App Commit Id:                b877d29
Gollum Android Lib Version:   0.7.17-pub-release
Gollum Android Lib Commit Id: b877d29
Gollum Device Lib Version:    0.0.7-pub-release
Gollum Device Lib Commit Id:  e340293
Gollum Device Lib Flavor:     e340293

Phone Information:
serial:                    TA9290DPBV
bootloader:                0x411A
manufacturer:              motorola
brand:                     motorola
hardware:                  qcom
board:                     MSM8226
device:                    falcon_umts
model:                     XT1032
display:                   LPB23.13-56
id:                        LPB23.13-56
type:                      user
user:                      hudsoncm
host:                      ilclbld36
fingerprint:               motorola/falcon_retgb/falcon_umts:5.1/LPB23.13-56/60:user/release-keys
tags:                      release-keys
time:                      1439854829000
radio firmware:            MSM8626BP_1032.3116.98.00R EMEA_CUST
version.codename:          REL
version.release:           5.1
version.incremental:       60
version.sdk_int:           22
Hash: ded54cd88a17ecfa228df53a3f02c4bb

ryan-dsouza, 2017-10-25:
The reset button in super cool by the way!

ryan-dsouza, 2017-10-29:
Any clues from the log @trollwookiee ?

trollwookiee, 2017-10-30:
I am trying to reproduce on a Moto G, Android 5.1 now.
Will keep you posted.
What is the fail rate ? Do you manage to get some data or never works ?

ryan-dsouza, 2017-10-30:
It worked for a few times on a Moto G, Android 4.4 - Until I ran into this error. Having the same error on 2 hardware units.
I updated the Android version 5.1 on the Moto G to see if it was an Android related issue. Still no luck.
I have tried it with a new Google Pixel, with Android 7.1 - Same problem.

ryan-dsouza, 2017-10-30:
Is it possible to download an older version of the Android app ?

trollwookiee, 2017-10-31:
Do you remember on what version it used to work ?

ryan-dsouza, 2017-10-31:
One before the 20th October one.

trollwookiee, 2017-10-31:
I just sent you an email with v1.2.53 from Oct 17.
Please check if it works better.

ryan-dsouza, 2017-10-31:
Hi @trollwookiee I tried downloading the latest app released on 27th - I am able to read the registers do RX/TX. So there definitely seems to be an issue with 1.2.54 (Oct 20th).

trollwookiee, 2017-10-31:
latest app released on 27th or 17th ?

- v1.2.55 has been release **27th** Oct
- v1.2.53 has been release **17th** Oct

v1.2.53 is ok ?

ryan-dsouza, 2017-10-31:
I didnt actually receive your email with 1.2.53. I tried on the latest one 1.2.55 (On both Moto G and Google Pixel)

trollwookiee, 2017-10-31:
I used email you gave us when you ordered on pandwarf.com

trollwookiee, 2017-10-31:
anyway thanks for feedback.
Enjoy your new toy :-)

ryan-dsouza, 2017-10-31:
oh ok. Got it - It was in spam. Will test it and give feedback

ryan-dsouza, 2017-10-31:
Hi @trollwookiee  - Seems like this issue is intermittent now. I tried reflashing the hardware via support recovery and it works the first time around with the latest firmware and android app.

Whenever RX setup and RX works, it displays the C1111 bootloader version in FW menu. After few times of receiving the signals,  RX stops working. The C1111 bootloader version is usually displayed as Unknown in those circumstances ( Attached screenshots and debug trace). Might be easier to read the debug trace from the reverse . 

Any help is appreciated. 
--------------------------------

PandwaRF Device Information:
MAC Address:                  F2:14:AB:37:92:D9
FW CC1111 Commit Id:          
FW CC1111 Version:            1.0.17
FW Nordic Commit Id:          
FW Nordic Version:            0.9.17

PandwaRF App Information:
APPLICATION_ID:               com.comthings.pandwarf
BUILD_TYPE:                   release
FLAVOR:                       pub
VERSION_CODE:                 10255
VERSION_NAME:                 1.2.55
App Commit Id:                cb47233
Gollum Android Lib Version:   0.7.18-pub-release
Gollum Android Lib Commit Id: cb47233
Gollum Device Lib Version:    0.0.7-pub-release
Gollum Device Lib Commit Id:  4423715
Gollum Device Lib Flavor:     4423715

Phone Information:
serial:                    TA9290DPBV
bootloader:                0x411A
manufacturer:              motorola
brand:                     motorola
hardware:                  qcom
board:                     MSM8226
device:                    falcon_umts
model:                     XT1032
display:                   LPB23.13-56
id:                        LPB23.13-56
type:                      user
user:                      hudsoncm
host:                      ilclbld36
fingerprint:               motorola/falcon_retgb/falcon_umts:5.1/LPB23.13-56/60:user/release-keys
tags:                      release-keys
time:                      1439854829000
radio firmware:            MSM8626BP_1032.3116.98.00R EMEA_CUST
version.codename:          REL
version.release:           5.1
version.incremental:       60
version.sdk_int:           22
Hash: 0107212373046f49d3cd8371af8d8550
 
[logcat.zip](https://github.com/ComThings/PandwaRF/files/1431475/logcat.zip)

[image]
[image]

trollwookiee, 2017-11-08:
Shall be fixed in Android V1.2.57.

closed 2017-11-08 by trollwookiee

ryan-dsouza, 2017-11-08:
Many thanks !