ComThings/PandwaRF #28: PandwaRF stops rx/tx from android app

TadasSu, 2017-10-31:
PandwaRF stops receiving/transmitting frequently (every 5 - 20 time random). Registers also can not be read when this happens. I need to reconnect (disconnect/connect). Sometimes disconnect and turning BT off does not help - blue light always on, so I need to press reset on the unit.
No errors in BLE log.

TadasSu, 2017-10-31:
p.s. spectrum analyzer also does not work when this happens. I only can read device info in Bus service tab.
Firmware 0.9.19 / 1.0.17

ryan-dsouza, 2017-11-01:
Sounds like a similar issue to #26

TadasSu, 2017-11-01:
It always works for the fist time when connected, but yes, it sounds *almost* similar like #26 issue.

Sikloz, 2017-11-05:
Same issue as above RX/TX sniffing/transmitting not working nor is the spec analyser but SA works on initial startup of the app.

trollwookiee, 2017-11-09:
Fixed in V1.2.58

closed 2017-11-09 by trollwookiee