ComThings/PandwaRF #31: USB doesnt work with Android App

ryan-dsouza, 2017-11-11:
Trace attached. Followed the sequence
1. Open App with BLE OFF
2. Connect USB reader
3. No functions work
4. Switched on BLE and connected to device (to get FW versions in logs)
....

PandwaRF Device Information:
MAC Address:                  FB:35:B3:12:A8:69
Model:                    PandwaRF Big
HW revision:            E
CC1111 Build Type:            GollumRfBigCCtl
CC1111 Bootloader Version:    CCTL v2
FW CC1111 Commit Id:          
FW CC1111 Version:            1.0.17
FW Nordic Commit Id:          
FW Nordic Version:            0.9.19

Rogue:                    false
Rogue Nordic:                 false
Rogue CC1111:                 false

PandwaRF App Information:
APPLICATION_ID:               com.comthings.pandwarf
BUILD_TYPE:                   release
FLAVOR:                       pub
VERSION_CODE:                 10258
VERSION_NAME:                 1.2.58
App Commit Id:                5db17dc
Gollum Android Lib Version:   0.7.20-pub-release
Gollum Android Lib Commit Id: 5db17dc
Gollum Device Lib Version:    0.0.9-pub-release
Gollum Device Lib Commit Id:  5db17dc
Gollum Device Lib Flavor:     5db17dc

Phone Information:
serial:                    881f8509
bootloader:                I9505XXUHOA7
manufacturer:              samsung
brand:                     samsung
hardware:                  qcom
board:                     MSM8960
device:                    jflte
model:                     GT-I9505
display:                   LRX22C.I9505XXUHOA7
id:                        LRX22C
type:                      user
user:                      dpi
host:                      SWDD6212
fingerprint:               samsung/jfltexx/jflte:5.0.1/LRX22C/I9505XXUHOA7:user/release-keys
tags:                      release-keys
time:                      1422470526000
radio firmware:            I9505XXUHOA7
version.codename:          REL
version.release:           5.0.1
version.incremental:       I9505XXUHOA7
version.sdk_int:           21
Hash: db6b6d2548c91c04f2255540ebb8cd15

Screen shots
[image]

[logcat.zip](https://github.com/ComThings/PandwaRF/files/1464111/logcat.zip)

trollwookiee, 2017-11-13:
did you enable USB from PandwaRF app ?

ryan-dsouza, 2017-11-13:
Had only this pop up? Couldnt find an option any where else to enable USB. Is there something I'm missing?

[image]

trollwookiee, 2017-11-13:
You need to enable dev mode. Then in bus service page, enable usb.
Remember that usb bypasses the nordic chip, so only RF features are active. Try cc1111 register read to check if usb works.

ryan-dsouza, 2017-11-14:
Hi @trollwookiee - No success yet.

To access the dev mode, I need to connect via bluetooth.  I connect via bluetooth and in the bus service page, enable USB. I'm able to read the C1111 Registers via Bluetooth.

I then disconnect bluetooth, while the device is plugged in via USB.Once the bluetooth is disconnected, I'm unable to communicate with the device. The read registers page is blanked out.

I connected the device back via bluetooth. It has lost the enable USB setting in the bus services page.

closed 2018-06-14 by trollwookiee