ComThings/PandwaRF #35: Incorrect dependencies in Android SDK 0.7.21

ryan-dsouza, 2017-11-23:
SDK doesnt compile.

Incorrect dependency
<dependency>
<groupId>AndroidGollum</groupId>
<artifactId>nrftoolbox</artifactId>
<version>unspecified</version>
</dependency>

ryan-dsouza, 2017-11-24:
Think the version needs to be specified.
Here is the error.

Error:Could not resolve all files for configuration ':app:debugCompileClasspath'.
> Could not find AndroidGollum:nrftoolbox:unspecified.
  Searched in the following locations:
      file:/Users/xxxx/Library/Android/sdk/extras/m2repository/AndroidGollum/nrftoolbox/unspecified/nrftoolbox-unspecified.pom
      file:/Users/xxxx/Library/Android/sdk/extras/m2repository/AndroidGollum/nrftoolbox/unspecified/nrftoolbox-unspecified.jar
      file:/Users/xxxx/Library/Android/sdk/extras/google/m2repository/AndroidGollum/nrftoolbox/unspecified/nrftoolbox-unspecified.pom
      file:/Users/xxxx/Library/Android/sdk/extras/google/m2repository/AndroidGollum/nrftoolbox/unspecified/nrftoolbox-unspecified.jar
      file:/Users/xxxx/Library/Android/sdk/extras/android/m2repository/AndroidGollum/nrftoolbox/unspecified/nrftoolbox-unspecified.pom
      file:/Users/xxxx/Library/Android/sdk/extras/android/m2repository/AndroidGollum/nrftoolbox/unspecified/nrftoolbox-unspecified.jar
      file:/Applications/Android Studio.app/Contents/gradle/m2repository/AndroidGollum/nrftoolbox/unspecified/nrftoolbox-unspecified.pom
      file:/Applications/Android Studio.app/Contents/gradle/m2repository/AndroidGollum/nrftoolbox/unspecified/nrftoolbox-unspecified.jar
      https://dl.google.com/dl/android/maven2/AndroidGollum/nrftoolbox/unspecified/nrftoolbox-unspecified.pom
      https://dl.google.com/dl/android/maven2/AndroidGollum/nrftoolbox/unspecified/nrftoolbox-unspecified.jar
      https://jcenter.bintray.com/AndroidGollum/nrftoolbox/unspecified/nrftoolbox-unspecified.pom
      https://jcenter.bintray.com/AndroidGollum/nrftoolbox/unspecified/nrftoolbox-unspecified.jar
      http://mymavenrepo.com/repo/l8uzsKQuWwYGByPmlbZ4/AndroidGollum/nrftoolbox/unspecified/nrftoolbox-unspecified.pom
      http://mymavenrepo.com/repo/l8uzsKQuWwYGByPmlbZ4/AndroidGollum/nrftoolbox/unspecified/nrftoolbox-unspecified.jar
  Required by:
      project :app > com.comthings.pandwarf:gollum:0.7.21

trollwookiee, 2017-11-24:
Fixed in SDK 0.7.22
Use:
dependencies {
    compile 'com.comthings.pandwarf:gollum:0.7.22'
...
}

closed 2017-11-24 by trollwookiee