ComThings/PandwaRF #42: Battery level indication after a battery reconnect

J-Run, 2018-01-09:
After a battery reconnection charge indicator on Bus Service tab of Android app always shows Level: 0%
Very annoying in case of adding power switch into a cut of powerline to be able of total powering off device.

Steps to reproduce:

1. disconnect device from USB
2. Disconnect internal battery from device (desolder or kind of)
3. Connect battery back
4. Connect to device via app
5. push Reload button on Bus Service tab

to fix it back:

1. Disconnect internal battery from device
2. Power on device with USB plug
3. Connect battery back
4. disconnect device from USB

While battery connected, indicator values are correct.

trollwookiee, 2018-01-09:
At power on, we assume we were previously shut down because of empty battery, so we update the Battery Gas Gauge to 0. This is expected behaviour, as I didn't considered the "battery desolder" case.
Will change behaviour in next release.

trollwookiee, 2018-06-12:
Fixed in Nordic FW 0.10.3

closed 2018-06-12 by trollwookiee