ComThings/PandwaRF #44: Issues with Mediatek chipsets

trollwookiee, 2018-01-17:
Extract from http://pandwarf.boards.net/thread/27/app-work-on-different-phones

2 phones tested, nothings works or near.

- On Android 7.0 phone (Doogee S60) :
Everything works as it should at the beginning, scanning, connect, display informations about hardware firmware etc.

(I don't have paired any of them with the PandwaRF, nothing in the Android's BT menu listed).

But when I try the spectrum analyser, the app become little weird : the title of the APP "PandwaRF" change from white to red colour, then a few seconds after it start to display some white noise and I never see any 433MHZ data's from a remote. Pressing stop / start didn't change anything. Need to kill the app to scan and connect again. Note : until I do force-close the APP, the blue LED is still OK, means the BLE link seems OK ?

In the Tx/Rx menu, if I click on "Measure", most of time nothing happened, no data seen and need to force-close (and title of APP change to WHITE/RED flipping randomly). But sometime I have the green LED acting in rhythm with the remote, but no data on APP. I was lucky one or two time with the "Sniff" function and see some datas onscreen, but I need to try & force-close 10 or 15 time before seeing something. If I increase the frame to 128 I lost the BLE link.

Another bug, when I activate developer functions, goes to CC1111RF menu and "Read all" registers, it mostly failed to display them and I can't try more than 10 times, the app crash and ask for sending a bug report.

- On the Android 6.0 (NOMU S30) :  Everything works as bad, except one thing : the "Read all" for CC1111 registers works like a charm ! No crash, values OK each time.

Cf Thread in forum
http://pandwarf.boards.net/thread/27/app-work-on-different-phones

Phil242, 2018-01-18:
Hi Trollwookiee,

As said on tweeter, I can send you the NOMU S30, this is a spare phone "in case of".
Not sure debuging the app on this one will solve ALL the Mediatek troubles, but it's a good start.

Just tell me,

Phil

trollwookiee, 2018-01-29:
Thanks Phil242, this is very kind.
I am working on Honor 6X right now. Will keep posted on the results and if additional test is required

Phil242, 2018-01-29:
Hi, nice to heard this good news. 

If you need a test of a beta APK, just ask, no problem for doing a report. I have here a few Chinese 433MHZ static alarm remote, 433MHZ rolling-code car keys etc.

Phil

trollwookiee, 2018-02-23:
Shall be fixed in Android v1.4.4

closed 2018-02-23 by trollwookiee