ComThings/PandwaRF #46: Detect carrier frequency problems

ellenhill94, 2018-03-29:
I am trying to use the PandwaRF to detect the base frequency of signals in the range 433.5 MHz to 434.5 MHz. I have used the API for Android, and have used rfSpecanGetRssi to get RSSI values in this range over a period of 5 seconds (refresh rate 30ms). Then I calculate the frequency based on the RSSI values retrieved. 
However, this method is not working well as the RSSI values are not very accurate, especially if I use an antenna. Has anybody had success with any other methods for obtaining the base frequency, or have any ideas for a better method?
Thanks in advance!

trollwookiee, 2018-04-04:
Do you think an API to get the frequency with the highest RSSI directly from the HW would be better ?

ellenhill94, 2018-04-04:
Yes definitely worth a try!

closed 2019-08-02 by trollwookiee