ComThings/PandwaRF #47: CC1111_NPI: Send synch data timed out

ellenhill94, 2018-04-09:
Every ~10 times RxSetup() or RxStop() is called (android API), the PandwaRF device crashes. The error is:

E/    CC1111_NPI: Send synch data timed out
E/   CC1111_RF_CONTROL: cc1111_ReadReg: Error, peek returned null
V/  CC1111_RF: rfsub1g_cc1111_Rf_GetRadioState: Retrieved radio state: 00: SLEEP
E/  CC1111_RF: rfsub1g_cc1111_Rf_SetRfMode: SetRfMode Failed due to a timeout of 400 us
E/  CC1111_RF: rfsub1g_cc1111_Rf_StartRx: Failed to start RX
E/CTCOM_RF_RXTX: ctcom_rx_setup: Failed with freq: 433930000, mod: 0, drate: 1440, flen: 100, chanbw: 100000, deviation: 20000

Any idea how to mitigate this?

trollwookiee, 2018-04-10:
what is the Gollum Android Lib version used ?

also there are 2 rxSetup() methods:
- public int rxSetup(final int frequency, final int modulation, final int dataRate, final int frameLength, final int chanbw, final int deviation) ==>  **It needs to be called from a background task**
- public void rxSetup(final int frequency, final int modulation, final int dataRate, final int frameLength, final int chanbw, final int deviation, **final GollumCallbackGetInteger cb**)  ==> This ones creates an an AsyncTask to run safely.

Same for rxStop().

Can you check you always make rxSetup() and rxStop() call from background ?

ellenhill94, 2018-04-10:
Gollum Android Lib is 0.9.3
Both functions are called from a background task.

closed 2018-10-23 by trollwookiee