ComThings/PandwaRF #48: rxFlush() and flushMsgQueue() do not remove data in buffer overflow

ellenhill94, 2018-04-09:
I want to remove any overflow from the receive buffer after the requested buffer size has been received. At the moment I am doing this using RxStop() followed by RxSetup(), which refreshes the receive buffer, but I want to keep the dongle in Rx mode and flush the buffer overflow so that, the next time I receive data, there is no data waiting in the next buffer. rxFlush and flushMsgQueue do not help with this. Maybe they were not made to do this - if so, is there another function that could perform this task?

trollwookiee, 2018-04-10:
These are the correct functions. But looking at the code, there are not correct.
I will fix this in next release.

trollwookiee, 2018-04-10:
Update: rxFlush() is wrong (fixed in next release) but flushMsgQueue() shall work incurrent API release.
Can you paste a log of what happens with flushMsgQueue() ?

ellenhill94, 2018-04-10:
Thanks. Sure, the flushMsgQueue log is as follows:

V/ CC1111_API: rfsub1g_msg_queue_flush:
D/    CC1111_NPI: npi_freeQueues: freeing all queues
D/    CC1111_NPI: npi_freeQueue: freeing 0xd04a9010 queue
D/Event: No subscribers registered for event class com.comthings.gollum.api.gollumandroidlib.events.LogDataEvent
V/     SLIST: Enter slist_remove_all
V/    CC1111_NPI: 0 element(s) in linked list after npi_freeQueue
V/    CC1111_NPI: Freed 13 element(s) from linked list
D/Event: No subscribers registered for event class de.greenrobot.event.NoSubscriberEvent
W/GollumDongle: Overwriting GollumBleManagerCallbacks
D/Event: No subscribers registered for event class com.comthings.gollum.api.gollumandroidlib.events.LogDataEvent
D/Event: No subscribers registered for event class de.greenrobot.event.NoSubscriberEvent
D/BleProfileServiceReady: setBleManagerCallbacks: com.comthings.gollum.api.gollumandroidlib.GollumDongle$10@874ba9e

trollwookiee, 2018-06-29:
It shall have been fixed in 0.9.8. Can you check on your side ?

closed 2018-07-05 by trollwookiee