ComThings/PandwaRF #49: Link to button

sjames02, 2018-04-22:
Hi,

I own a Rogue Pro, unless I am misunderstanding the 'link to button' feature, I don't believe it is working for me.

After capturing the data from an OOK transmitter and then confirming successful capture by re-transmitting it, I then try to link the configuration/data to a physical button. Upon pressing the hardware button (short press) i do get a red LED, however it does not appear to be transmitting anything. I have tried pressing the button from within 'Bus Service Extended' of which also presents a red LED with no apparent transmission?

Appreciated.

trollwookiee, 2018-04-22:
Hi,

What HW revision do you have ?

sjames02, 2018-04-22:
Its E4, Rogue Pro. I purchased just a couple of weeks ago. Firmware is up to date, 1.0.22.

trollwookiee, 2018-04-26:
fix is ongoing.

trollwookiee, 2018-06-14:
Fixed in 1.4.8

closed 2018-06-14 by trollwookiee