ComThings/PandwaRF #55: updateFirmwareCC1111 using hex file not work

LSDsl, 2018-07-02:
Update firmware using [updateFirmwareCC1111](https://comthings.github.io/PandwaRF/v0.9.10/com/comthings/gollum/api/gollumandroidlib/GollumDongle.html#updateFirmwareCC1111-java.lang.String-com.comthings.gollum.api.gollumandroidlib.callback.GollumCallbackGetInteger-) not work.

After call updateFirmwareCC1111 it immediately call callback (GollumCallbackGetInteger ) with 0 and no any exception. And not update CC1111.

trollwookiee, 2018-07-05:
Fixed in 0.9.10

closed 2018-07-05 by trollwookiee

LSDsl, 2018-07-05:
Tested on 0.9.10 - not work.

trollwookiee, 2018-07-05:
Sorry typo, 0.9.10 is current version. Next one is 0.9.11. Updated Milestone.

trollwookiee, 2018-07-06:
can you check the lib 0.9.11 ? https://comthings.github.io/PandwaRF/

LSDsl, 2018-07-07:
Also I have question about cbOnProgress listener. In my test it called with parameters (pages) from 2 to 28. Why it starts from 2 and ends with 28 ?? Max will be 28 always? After 28 was cbOnFinish with exception as I described above.

trollwookiee, 2018-07-09:
1) What is the error type you get in cbOnFinish() ?
2) cbOnProgress listener: 2 to 28 is normal. I will describe more in wiki.

LSDsl, 2018-07-09:
There is no any error type - only exception "Gollum firmware update failed"  (look above) https://comthings.github.io/PandwaRF/v0.9.11/com/comthings/gollum/api/gollumandroidlib/callback/GollumCallback.html

trollwookiee, 2018-07-09:
Ok, this spurious call to cbOnFinish() is fixed now. Thanks for report.

LSDsl, 2018-07-19:
> Pages generally range from 1 to N=30, but this may vary according to FW size.
I need to know current page and max page count. But how I can know max N for updating FW?

I need this to show progress like in PandwaRF app.
[image]

trollwookiee, 2018-07-19:
https://comthings.github.io/PandwaRF/v0.9.12/index.html
When called, the hex file is analyzed and callback cbOnPageTotalUpdate is called with total number of pages to be flashed (N).
N is max page count.
cbOnProgress contains current page.

LSDsl, 2018-08-06:
But now onSoftwareVersionReceived onFirmwareVersionReceived work incorrect - bug https://github.com/ComThings/PandwaRF/issues/60

LSDsl, 2018-08-06:
Also connection time increased in last sdk and in log many errors like

> W/System: Ignoring header X-Parse-Client-Key because its value was null.
>           Ignoring header X-Parse-Application-Id because its value was null.