ComThings/PandwaRF #58: Feature request: Writing captured datas to phone storage.

PsychoBitchy, 2018-07-12:
I would like to be able to save captured datas to phone storage ( other than memory slot or post to online API )
For example, when listening to rolling code challenges / response, I would like collect a lenghty sum of key echanges for analysis.
Thanks.

ferprat, 2018-11-29:
I also think it could be very useful, Having the "buttons" saved in a txt or something like that automatically in an Android file.

Also, it would be great if it was possible to copy/paste the sniffed data. I don't understand why it is locked (I know I can write or erase, but I would like to SELECT it).

trollwookiee, 2019-05-23:
Save to JSON is being generalized right now. Target is next release (Android 1.5.5)

closed 2019-05-23 by trollwookiee