ComThings/PandwaRF #60: onSoftwareVersionReceived /onFirmwareVersionReceived received incorrect values [SDK 0.9.12]

LSDsl, 2018-08-06:
Callback onFirmwareVersionReceived / onSoftwareVersionReceived  received  incorrect values:
onSoftwareVersionReceived - 46a8b34
onFirmwareVersionReceived - 94e710c

Must be something like:
onSoftwareVersionReceived - 0.10.3
onFirmwareVersionReceived - 1.0.24

trollwookiee, 2018-08-06:
Any logcat error or info associated ?
Mapping between Commit Id and version is managed on servers.
Commit Id (94e710c) is returned instead of version (1.0.24) when there is no server connection or commit Id doesn't exist on server.
Do you have disabled backend connection ?(https://comthings.github.io/PandwaRF/v0.9.12/com/comthings/gollum/api/gollumandroidlib/GollumDongle.html#setOfflineModeBackend-boolean-)

LSDsl, 2018-08-06:
No. not disabled backend connection. Also when rollback to 0.9.10- all work ok.

LSDsl, 2018-08-06:
```I/BleManager: Read Response received from 00002a27-0000-1000-8000-00805f9b34fb, value: (0x) 45
I/BleManager: nextRequest: [4] Request unqueued: READ, characteristic: 00002a28-0000-1000-8000-00805f9b34fb, value: 
              Reading characteristic 00002a28-0000-1000-8000-00805f9b34fb
D/BleManager: readCharacteristic : true
I/BleManager: Read Response received from 00002a28-0000-1000-8000-00805f9b34fb, value: (0x) 34-36-61-38-62-33-34
V/GollumParse: readFirmwareNordicVersion, commitID: 46a8b34
D/Event: No subscribers registered for event class com.comthings.gollum.api.gollumandroidlib.GollumParse$a
         No subscribers registered for event class de.greenrobot.event.NoSubscriberEvent
I/BleManager: nextRequest: [3] Request unqueued: READ, characteristic: 00002a26-0000-1000-8000-00805f9b34fb, value: 
              Reading characteristic 00002a26-0000-1000-8000-00805f9b34fb
D/BleManager: readCharacteristic : true
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
I/BleManager: Read Response received from 00002a26-0000-1000-8000-00805f9b34fb, value: (0x) 39-34-65-37-31-30-63
V/GollumParse: readFirmwareCC1111Version, commitID: 94e710c
D/Event: No subscribers registered for event class com.comthings.gollum.api.gollumandroidlib.GollumParse$a
         No subscribers registered for event class de.greenrobot.event.NoSubscriberEvent
I/BleManager: nextRequest: [2] Request unqueued: READ, characteristic: dead1527-2dbb-4d90-91d7-bdc47b265643, value: 
              Reading characteristic dead1527-2dbb-4d90-91d7-bdc47b265643
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
D/BleManager: readCharacteristic : true
I/BleManager: Read Response received from dead1527-2dbb-4d90-91d7-bdc47b265643, value: (0x) 00-00-00-D0
                  onBusConfigUsbAllowed true
I/BleManager: nextRequest: [1] Request unqueued: READ_CCCD_DESCRIPTOR, characteristic: dead1524-2dbb-4d90-91d7-bdc47b265643, value: 
              Reading CCCD descriptor dead1524-2dbb-4d90-91d7-bdc47b265643
D/BleManager: readCccdDescriptor : true
D/BleManager: onDescriptorRead 00002902-0000-1000-8000-00805f9b34fb
I/BleManager: nextRequest: [0] Request unqueued: READ_CCCD_DESCRIPTOR, characteristic: 00002a19-0000-1000-8000-00805f9b34fb, value: 
              Reading CCCD descriptor 00002a19-0000-1000-8000-00805f9b34fb
D/BleManager: readCccdDescriptor : true
D/BleManager: onDescriptorRead 00002902-0000-1000-8000-00805f9b34fb
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
D/Event: No subscribers registered for event class com.comthings.gollum.api.gollumandroidlib.GollumParse$a
         No subscribers registered for event class de.greenrobot.event.NoSubscriberEvent
I/GollumParse: readFirmwareNordicVersion complete:0
V/GollumParse: updateInfoFirmwareNordic, installationId: 2209b1b1-3f03-4717-aebf-145c30a421f2
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
D/Event: No subscribers registered for event class com.comthings.gollum.api.gollumandroidlib.GollumParse$a
         No subscribers registered for event class de.greenrobot.event.NoSubscriberEvent
I/GollumParse: readFirmwareCC1111Version with commitID complete: 0 FW found
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
[... cut ...]
          Ignoring header X-Parse-Application-Id because its value was null.
I/GollumParse: i/o failure
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
W/System: Ignoring header X-Parse-Client-Key because its value was null.
W/System: Ignoring header X-Parse-Application-Id because its value was null.
W/System: Ignoring header X-Parse-Client-Key because its value was null.
          Ignoring header X-Parse-Application-Id because its value was null.
```

trollwookiee, 2018-08-10:
Mapping between CommitId and version is indicated in each release description in https://github.com/ComThings/PandwaRF/wiki/Firmware-Releases-Nordic and https://github.com/ComThings/PandwaRF/wiki/Firmware-Releases-CC1111

closed 2018-08-10 by trollwookiee