ComThings/PandwaRF #62: RX power amplifier option does not work

TadasSu, 2018-10-04:
RSSI in specan is the same with different RF Power Amplifiers options.
I tried these: Turn off TX/RX amp / Turn on RX amp always / Turn on RX apm only when receiving / Turn on RX/TX auto

[image]
[image]
[image]

trollwookiee, 2018-11-15:
I made some tests on App v1.4.20/Nordic v0.11.1/CC1111 v1.1.1.
I measure a difference of 9 dB when RF power amplifiers are ON in RX. 

- With RX amplifiers OFF: RSSI = **-30dBm** at 433.4MHz
[image]

- With RX amplifiers Always ON: RSSI = **-21dBm** at 433.375MHz
[image]

- With RX amplifiers in Auto mode: RSSI = **-21dBm** at 433.4MHz
[image]

So I assume that RX power amplifier option does work.

Please re-open in case I am wrong.
Thanks.

closed 2018-11-15 by trollwookiee

TadasSu, 2018-11-24:
I can not reproduce issue on latest firmware too. It was on older one.