ComThings/PandwaRF #66: Sync Word does not work on 3bit encoding

TadasSu, 2018-11-01:
Sync Word does not work on 3bit encoding. For example I can not get sync word 0000FF to work. URH show random spikes, but not any of 0's or 1's

trollwookiee, 2018-11-02:
I noticed an issue, maybe not linked to this one: there is an empty space between synchro and codeword corresponding to 1 byte of data. But this is regardless of encoding base (occurs in base 2/3/4). Not sure this is what you are refering to...

I have opened https://github.com/ComThings/PandwaRF/issues/70 to track

trollwookiee, 2018-11-02:
Issue #70 is fixed, but I need more detail for this issue to know if it is related. A screenshot showing paramters would be great

TadasSu, 2018-11-02:
I have these when I use HT12E template for 3bit input:
[image]
[image]
[image]

I can not input 3bit on other templates (or without template).
Relates to #71

TadasSu, 2018-11-02:
update:
If I set sync word in json template, it works. It does not work only if used without custom template.

trollwookiee, 2018-11-05:
I can't reproduce. No spikes at all on Nordic v0.11.1 and CC1111 v1.1.1. What is your FW versions for Nordic and CC1111 ?

TadasSu, 2018-11-05:
Nordic: 0.11.1
CC1111: 1.0.24

I can reproduce it every time:
Select model HT12E_A10D2 template.
Change Mask to all F's
Change Value to all 0's
Change SyncWord to 0000FF

trollwookiee, 2018-11-05:
Please update CC1111 version to 1.1.1.

TadasSu, 2018-11-05:
> Last CC1111 FW Version: Unknown
 in FW tab.

trollwookiee, 2018-11-05:
CC1111 FW update issue shall be fixed. Please try updating again.

TadasSu, 2018-11-05:
FW updated successfully.
I will try to reproduce issue with new fw later.

TadasSu, 2018-11-06:
I can reproduce issue with new fw too.
(Data Rate: **25 600**)

trollwookiee, 2020-08-05:
Fixed (a long time ago)

closed 2020-09-21 by trollwookiee