ComThings/PandwaRF #76: Sync failed in Android Studio

Massimaux, 2018-11-17:
Hi,

I downloaded the source code from https://github.com/ComThings/PandwaRF/tree/master/SW/Android/SampleApps/PandwarfSDK and opened it in Android Studio. Unfortunately, the sync procedure failed and I was not able to build apk from your sample codes. 

What could be the problem?

trollwookiee, 2018-11-20:
I just tested against gollum:0.9.9 and gollum:0.9.12 and both sync and compile worked.

Can you describe more precisely what kind of sync issue you face ?

Massimaux, 2018-11-26:
Hi,

There is no sync issue anymore. This is the status with respect to its installation and working on different smartphones:

LG K8 (Android 6.0 | API level 23): WORKING
HUAWEI P10 Lite (Android 8.0 | API level 26): CONSTANT CRASHING ON LAUNCH

Massimaux, 2018-11-26:
I will try to obtain it later today.

BTW, the playstore PandwaRF app works perfectly on this HUAWEI P10 Lite.

Massimaux, 2018-11-26:
> do you have a log for HUAWEI P10 Lite crash ?

Here is the log reporting the crash of sample PandwaRFSdk:

[Huawei P10 Lite Crash Log.txt](https://github.com/ComThings/PandwaRF/files/2617562/Huawei.P10.Lite.Crash.Log.txt)

Massimaux, 2018-11-26:
> do you have a log for HUAWEI P10 Lite crash ?

Here is the log reporting the crash of sample PandwaRFSdk:

[Huawei P10 Lite Crash Log.txt](https://github.com/ComThings/PandwaRF/files/2617714/Huawei.P10.Lite.Crash.Log.txt)

trollwookiee, 2018-11-27:
`  
    --------- beginning of crash
2018-11-26 23:31:10.813 25466-25466/com.comthings.pandwarf.sample.PandwaRfSdk.debug E/AndroidRuntime: FATAL EXCEPTION: main
Process: com.comthings.pandwarf.sample.PandwaRfSdk.debug, PID: 25466    
java.lang.UnsatisfiedLinkError: dlopen failed: "/data/data/com.comthings.pandwarf.sample.PandwaRfSdk.debug/app_lib/libusb1.0.so" is 32-bit instead of 64-bit
`

Try to build your app with "armeabi-v7a"
In your build.gradle:

```
defaultConfig {
...
   ndk {
            abiFilters "armeabi-v7a"
        }
...
}

```

Massimaux, 2018-11-28:
Thank you! Crash solved.

If I have a data buffer of 15-20 Kbytes to be transmitted, how can I minimize the probability of Tx failure:
- by calling one txSend() once
Or
- by dividing the data into multiple blocks and calling multiple instances of txSend() ?

closed 2018-11-28 by Massimaux