ComThings/PandwaRF #77: Transmission does not work with USB connection

Massimaux, 2018-11-18:
Hi,

I've had the PandwaRF dongle for about 10 days. The USB connection for both Rx and Tx worked well with the original firmware.

Then, I upgraded the firmware as suggested by the PandwaRF application. After that only the BLE connection works. The USB does not work for transmissions. When I press Xmit, it gets stuck and the RED LED remains ON.

What happened? Is this problem caused by the firmware upgrade or is it something else?

Thank you

trollwookiee, 2018-11-19:
Can you precise from what version you upgraded from and to ?

trollwookiee, 2018-11-19:
Update: USB works also with LGE Nexus 5x, Android 8.1.0, API 27

trollwookiee, 2018-11-19:
In conclusion, I think USB works. I have noticed 2/3 minor issues - icon not updated, splash screen not displayed, reconnection in BLE at the same time as USB, etc... - that I will fix, but nothing really blocking.

Massimaux, 2018-11-19:
Thank you for the prompt responses.

Attached is the firmware configuration. I tried it with two identical phones (Huawei P10 Lite, Android 8.0.0) and the same problem happens on both phones.
[image]

Also, please have a look at the cable that I'm using for the USB connection. It's a concatenation of two USB cables so that the two ends are micro USB Type B. But the two ends of this cable (let's call them End1 and End2) do not seem to be equal. I noticed two different situations:
- when End1 is plugged in the PandwaRF, USB is detected and only Rx mode works
- when End2 is plugged in the PandwaRF, USB is not detected and nothing works
[image]

Perhaps this cable is not suitable for the USB connection. However, it used to work 10 days ago and that's what confuses me. :(

Massimaux, 2018-11-19:
Thank you for the prompt responses.

Attached is the firmware configuration. I tried it with two identical phones (Huawei P10 Lite, Android 8.0.0) and the same problem happens on both phones.
[image]

Also, please have a look at the cable that I'm using for the USB connection. It's a concatenation of two USB cables so that the two ends are micro USB Type B. But the two ends of this cable (let's call them End1 and End2) do not seem to be equal. I noticed two different situations:

when End1 is plugged in the PandwaRF, USB is detected and only Rx mode works
when End2 is plugged in the PandwaRF, USB is not detected and nothing works

Perhaps this cable is not suitable for the USB connection. However, it used to work 10 days ago and that's what confuses me. :(

[image]

trollwookiee, 2018-11-19:
I don't think the cable is correct (don't know why it worked though...)
Please check that the pinout is the same as the following diagram:
[image]

The **direction is important**:
- A device is the host (the smartphone)
- B device is the PandwaRF

Here is a cable that is guaranteed to work.
https://www.ebay.fr/itm/Micro-USB-male-to-Micro-USB-Male-data-charger-cable-for-MP4-Mobile-Phone/281892636170

Massimaux, 2018-11-19:
Good to know, thanks!

I already ordered this cable. Is that the right one or I made a mistake?

http://www.aliexpress.com/snapshot/0.html?spm=a2g0s.9042647.0.0.21d64c4dmCN6pX&orderId=96455028899404&productId=32808154910

Massimaux, 2018-11-19:
Good to know, thanks!

I already ordered this cable. Is that the right one or I made a mistake?

http://www.aliexpress.com/item/1pc-Micro-USB-Type-B-Male-To-Micro-B-Male-5-Pin-Converter-OTG-Adapter-Lead/32808154910.html?spm=a2g0s.12269583.0.0.17b5240dTBHQa7

trollwookiee, 2018-11-19:
I usually prefer cables with color to indicate which side is what, but it should be OK if product description is correct.

closed 2018-11-19 by trollwookiee

Massimaux, 2018-11-19:
Do you think this product description is OK?

Micro USB Type B Male To Micro B Male 5Pin Converter OTG Adapter Lead Data Cable

Compatible only with the Android Smartphones and Tablets having the USB Host OTG features

Closed #77.

trollwookiee, 2018-11-19:
Yes should be OK.

trollwookiee, 2018-11-21:
Here is a final summary of USB tests and fixes.
With Firmware Nordic v0.11.1, CC1111 v1.1.1, **USB works fine** with:

- Motorola MotoE2 on Android 6.0, API 23
- LGE Nexus 5x, Android 8.1.0, API 27   
- Samsung Galaxy S5 SM-G900T, Android 6.01.1, API 23
- Xiaomi Redmi 5, Android 7.1.2, API 25

**But some USB improvement/stability fixes will be released in Android App v1.4.21**

trollwookiee, 2018-11-27:
Updated https://github.com/ComThings/PandwaRF/wiki/Android-USB-Connection