ComThings/PandwaRF #81: Rogue reboots when data rate 20000

TadasSu, 2018-12-09:
Rogue reboots if data rate 20000 and payload 500.

TadasSu, 2018-12-10:
It seams that device reboots only if captured payload is bigger then desired payload (if I keep pressing button on keyfob)

trollwookiee, 2018-12-10:
Ok I see. BLE throughput is smaller than 20000 bits/s so PandwaRF fail to evacuate data to app.
This is clearly a known issue (at least internally), and I will try to address it shortly.
Things will get better when I will release RX frame compression for Rogue line.

trollwookiee, 2019-02-19:
Frame compression and data enqueuing are coming soon for the Rogue.
This will allow to fix crashes at high data rate.

trollwookiee, 2019-08-02:
New releases has been made yesterday:

- App version 1.6.0
- Nordic FW version 0.11.6
- CC1111 FW version 1.1.9

Please indicate if issue still occurs.

closed 2019-10-07 by trollwookiee