ComThings/PandwaRF #83: Specan GUI suggestions

TadasSu, 2018-12-11:
a) change start freq to center freq in specan. (Now if I want to scan 868.0 Mhz, I need to input 867.5 Mhz (then graph shows from 867.5 to 868.5, center freq 868.0))

b) change freq input box to dropdown box (with remembered values) like in tx/rx freq. (Maybe add some common frequencies by default, like 315, 417.90, 433.92, 434.42, 868.35, etc)

trollwookiee, 2018-12-17:
@TadasSu what bands would you like as default, except the one already mentioned ?

- 315
- 417.90
- 433-434
- 434-435
- 868-869
- 915
- ... ?

TadasSu, 2018-12-17:
You know better what are most common frequecies :)
It would be nice to get 434 center freq also (so 433.92 and 434.42 are in)

trollwookiee, 2018-12-17:
I wasn't aware of this 417.90MHz freq...

trollwookiee, 2019-01-17:
The following feature has been released:
> b) change freq input box to dropdown box

> a) change start freq to center freq in specan
Will come soon.

EvilDonkey420, 2019-02-13:
Would be nice to offer a DIY upgrade kit for the Pandwarf Big to rogue/pro . Would entice the maker market

trollwookiee, 2019-02-19:
@EvilDonkey420 the Rogue Pro has a different FW so DIY upgrade kit is not feasible.
And I think DIY/makers generally prefer low cost RF modules/arduino/etc...

trollwookiee, 2019-06-18:
Fixed in new SpecAn implementation

closed 2019-06-18 by trollwookiee