ComThings/PandwaRF #95: APP: Server: Not running

TadasSu, 2019-09-17:
The APP does not work correctly. I cant update pandwarf firmware, because app shows:
Platform: Online
Server: Not running

Problem occurred after app update.
Worked with v.1.6.0
NOT working with v1.6.1
NOT working with v1.6.2

With older app versions there was connections to http://gollumrf-parse-server.herokuapp.com/,
with v1.6.1 / v1.6.2 there are no connections to this url.

trollwookiee, 2019-09-17:
thanks.
it seems that it depends on location/operator.
Server is up and running.
I have tested with wifi and 4G and it works (at least from France).

Could you go to menu/About and send us a debug trace ?

TadasSu, 2019-09-17:
I have sent debug trace to you.
Maybe I can check if ip or url is accessible manually from my phone?

I do not think that this is location/operator issue, because I tried with 2 different mobile operators and 2 different cable operators. Also I have tried with VPN connection from another country.

trollwookiee, 2019-09-18:
No progress yet.
Still can't find out why it sometime fails.

The URL to check server status is a POST to:
http://gollumrf-parse-server.herokuapp.com/parse/functions/checkServer
with header:
```
Content-Type:application/json
X-Parse-Application-Id:9CmA2GkkROv4Q5t6mroAsCW1Zc5Hzisd08VhqvD5
```

result shall be :
```
{
    "result": true
}
```

Can you try ?

TadasSu, 2019-09-18:
I get {"result":true} from my mobile device with REST API application.

trollwookiee, 2019-09-20:
I think this issue appears only on Android 9 (API 28). I am investigating.

TadasSu, 2019-09-23:
Server shows online, but can not update fw after updating the app to newest ver.

[image]

trollwookiee, 2019-09-23:
With app 1.6.4 ?

TadasSu, 2019-09-23:
Yes. With 1.6.4-pub

trollwookiee, 2019-09-24:
Ok, another fox being tested today.
BTW this is probably caused by Android 9 policy for cleartext traffic: all communication shall now be using HTTPS.
If fix is successful, we will release it in 1.6.5.

trollwookiee, 2019-09-25:
I have ordered some more Android 9 phones to debug the issue (I have only 1 now)
In the meantime, I will release App 1.6.5 with the fix that is working on Honor 10 with Android 9.

TadasSu, 2019-09-25:
thanks for info

TadasSu, 2019-09-25:
I have update the app to the v1.6.5
It WORKS now! :)

trollwookiee, 2019-09-26:
Great !
Thanks for help closing this issue.

closed 2019-09-26 by trollwookiee