ComThings/PandwaRF #96: Marauda - Not detected as a Marauder

sjames02, 2019-10-05:
I own a 'Marauder Standard' and very frequently when trying to fetch data or configure a capture frequency the device fails to function. When I check the logs it reports "Feature not possible on this device. Requires a Marauder device."

The device has been updated and reports as a Marauder Standard within the information page. The application is also the newest available within the Google Play store.

Your help would be much appreciated.

trollwookiee, 2019-10-07:
New app release 0.0.5 + Nordic 0.11.10 + CC1111 1.1.13 are pushed today.
Please try again with these versions. Before updating the app, don't forget to save your RF captures.

trollwookiee, 2019-10-11:
Do you still have issue ?

trollwookiee, 2019-12-03:
Not able to reproduce the issue any more since app v0.0.5

closed 2019-12-03 by trollwookiee