ComThings/PandwaRF #97: Rx/Tx fragment won't let you specify a Frame length Less than 14 bytes

thesle3p, 2020-01-27:
Hi I am trying to sniff the  Simplisafe 2 protocol which uses 12 byte frames but the Rx/TX fragment won't let you set a frame length less than 14 bytes. 
[image]

trollwookiee, 2020-01-27:
This shall not be a problem, the RX frame length is only used internally for PandwaRF to know at which rate send the data to phone.
You can set a higher RX frame length.
The frame length counts in bytes as soon as a first byte is detected, so if a wakeup bit or a header is sent some time before the real data, if you set a 14 bytes frame length you will miss the interresting payload.
Also in your image, the RF amplifiers are _may be_ in the wrong state.  Click on the 3 dots in the upper right of the app to set the RF amplifiers

thesle3p, 2020-01-27:
You don't understand it won't let me send a length lower than 14. I need it to be 12 and the pandwaRF can't receive Simplisafe 2 packets with the RF amplifiers on OR off..... I have spent the past two months trying to get the PandwaRF to pick up a Simplisafe 2 PIN....

trollwookiee, 2020-01-28:
ok, so let me understand: the data you want to capture and which is sent over the air is only 12 bytes @ 2000bits/s, so it lasts 12*8/2000=48ms, right? No data/header before, no data after, only 12 bytes ?
If you do not receive enough data to complete the minimum 14 bytes required, you can always increase the sampling rate in the RX page. You will receive more data (oversampled).

closed 2020-03-02 by trollwookiee