ComThings/PandwaRF #98: New PandwaRF, Brute Force Attack Stops at 90%

xjcreeper, 2020-02-25:
We have a new PandwaRF RoguePro we have purchased for our lab testing. We have found that the Brute Force Attack Stops at 90%.
We have tested several FOBs so far and get the same result. Please contact us to help resolve this issue. Thank you kindly.

xjcreeper, 2020-02-25:
Checking FW it looks to be missing. I see that the BLE and RF firmware versions say 'unknown'. Please provide the firmware needed to update them.

trollwookiee, 2020-03-02:
Sometime the CC1111 FW is marked as Unknown, because the CC1111 chip fails to send to send its Version ID before the Nordic finishes its boot. It is not an issue. Generally a HW reset solves the issue. Also if brute force stops at 90%, it can be a mis-configuration of the BF params. Could you please send the BF parameter you used ?

closed 2020-08-05 by trollwookiee