ComThings/PandwaRF #101: Orange led does not blink when charging

Massimaux, 2020-06-18:
Hi,

I have used the pandwaRF for year and a half and it has worked very well up till today. The battery was low and I started charging the dongle. But, there was no orange led blinking. Also, the pandwaRF app cannot see it when scanning.

It has been charging for more than an hour but the only thing happening is short simultaneous orange and green blinks about every minute.

What is going on? How can I check if the dongle is still functional?

Thanks

trollwookiee, 2020-06-19:
Hi, 
few questions:

- is the USB power plugged into PandwaRF while scanning with the app ?
- when you connect the PandwaRF to your smartphone using USB cable, is it detected by the app ?
- a possibility is that the battery is dead

Massimaux, 2020-06-19:
> * is the USB power plugged into PandwaRF while scanning with the app ?

Yes. If the battery is dead and if PandwaRF is functional, is this case possible?

> * when you connect the PandwaRF to your smartphone using USB cable, is it detected by the app?

No.

Massimaux, 2020-07-02:
@trollwookiee 

If the battery is removed and the USB power is plugged in, can the PandwaRF work?

trollwookiee, 2020-07-02:
Yes, PandwaRF can work without battery

Massimaux, 2020-07-02:
> Yes, PandwaRF can work without battery

Thanks. Since the PandwaRF does not work when plugged with USB power, can we conclude that the dongle has failed?

trollwookiee, 2020-08-05:
If all LEDs are OFF when plugged in USB, then the board is probably dead