ComThings/PandwaRF #108: Kaiju - you are not logged

Cloud-Forge, 2021-07-10:
In the Android App, I connect the Rogue Pro to the APP.

I go to the Kaiju menu. 
(I have previously created a Kaiju account with a gmail address that is the one used on my Android device).

When I press the "LOGIN" button nothing happens.

dapollo, 2021-07-11:
It's the same problem I had a couple of days ago.
solution?
In the android application.
Go to about.
then sent debug trace with your kaijui account email.
pandwarf rogue administrators should enable it.

Cloud-Forge, 2021-07-11:
> It's the same problem I had a couple of days ago.
> solution?
> In the android application.
> Go to about.
> then sent debug trace with your kaijui account email.
> pandwarf rogue administrators should enable it.

Thanks Dapollo, will do that.

Rem193, 2021-07-12:
Hi @Cloud-Forge , your issue is under investigation, which email address do you use to connect to Kaiju ?

Rem193, 2021-11-19:
Hi,

Due to no more feedback provided, this issue is closed.
If you still have an issue with Kaiju login please open a new issue.

PandwaRF Team.

closed 2021-11-19 by Rem193