ComThings/PandwaRF #110: Rogue Pro with idProduct 0x60fe

wilecoyote2019, 2021-11-17:
Hi,
I i've a Pandwarf Rogue Pro and the product id is 0x60fe. 
To work with rfcat I had to modify this line in cc1111client.py : 

if (dev.idVendor == 0x0451 and dev.idProduct == 0x4715) or (dev.idVendor == 0x1d50 and (dev.idProduct == 0x6047 or dev.idProduct == 0x6048 or dev.idProduct == 0x605b or dev.idProduct == 0x60FF)):

f (dev.idVendor == 0x0451 and dev.idProduct == 0x4715) or (dev.idVendor == 0x1d50 and (dev.idProduct == 0x6047 or dev.idProduct == 0x6048 or dev.idProduct == 0x605b or dev.idProduct == 0x60FF or dev.idProduct == 0x60FE)):

If that is ok, can you add this update to git repository ?

Thanks,
Wil.e. Coyote

trollwookiee, 2021-11-19:
thanks for the finding. I am afraid a lot of messages are not up to date in our RfCat fork.
We need to update the RfCat script.

trollwookiee, 2022-02-08:
FYI, our RfCat fork update is ongoing... Shall be ready by end of week

trollwookiee, 2022-02-10:
RfCat updated in https://github.com/ComThings/rfcat

closed 2022-02-10 by trollwookiee